tinygo-org/bluetooth #253: Unable to compile for `pca10059` as a target

Schwusch, 2024-03-20:
The build target `pca10059` aka [nRF52840 Dongle](https://www.nordicsemi.com/Products/Development-hardware/nRF52840-Dongle) does not compile the [NUS server example](https://github.com/tinygo-org/bluetooth/blob/release/examples/nusserver/main.go). I get the following error:

```bash
tinygo build -target=pca10059 -o flash.hex ./main.go
# tinygo.org/x/bluetooth
../../go/pkg/mod/tinygo.org/x/bluetooth@v0.8.0/gap.go:80:10: undefined: Address
../../go/pkg/mod/tinygo.org/x/bluetooth@v0.8.0/gatts.go:16:10: undefined: Characteristic
../../go/pkg/mod/tinygo.org/x/bluetooth@v0.8.0/adapter.go:9:10: undefined: Adapter
../../go/pkg/mod/tinygo.org/x/bluetooth@v0.8.0/adapter.go:9:51: undefined: Address
```

deadprogram, 2024-03-21:
Hi @Schwusch looks that that board needs the softdevice added in TinyGo itself in similar way as the pca10056.

Please see https://github.com/tinygo-org/tinygo/blob/release/targets/pca10056-s140v7.json for example.

We would be happy to accept a PR that added it! :smile_cat:

deadprogram, 2024-12-12:
Closing since this was added some time ago. Thank you!

closed 2024-12-12 by deadprogram