tinygo-org/bluetooth #265: Ubuntu 22.04 Error AddService - Failed to create characteristic entry in database

Martim-Sensefinity, 2024-04-17:
**Bluetooth version:**
bluetoothctl: 5.75

**Version: 0.9.0**

**Ubuntu version:**
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy

**Logs**

```
Apr 17 20:13:15 martim-sensefinity bluetoothd[25936]: src/gatt-database.c:database_add_chrc() Failed to create characteristic entry in database
Apr 17 20:13:15 martim-sensefinity bluetoothd[25936]: src/gatt-database.c:database_add_service() Failed to add characteristic
Apr 17 20:13:15 martim-sensefinity bluetoothd[25936]: src/gatt-database.c:database_add_app() Failed to add service
Apr 17 20:13:15 martim-sensefinity bluetoothd[25936]: src/gatt-database.c:client_ready_cb() Failed to create GATT service entry in local database
```

**Code (same as example Nordic server)**

```go
must("add service", adapter.AddService(&bluetooth.Service{
		UUID: serviceUUID,
		Characteristics: []bluetooth.CharacteristicConfig{
			{
				Handle: &rxChar,
				UUID:   rxUUID,
				Flags:  bluetooth.CharacteristicWritePermission | bluetooth.CharacteristicWriteWithoutResponsePermission,
				WriteEvent: func(client bluetooth.Connection, offset int, value []byte) {
					txChar.Write(value)
					for _, c := range value {
						rawterm.Putchar(c)
					}
				},
			},
			{
				Handle: &txChar,
				UUID:   txUUID,
				Flags:  bluetooth.CharacteristicNotifyPermission | bluetooth.CharacteristicReadPermission,
			},
		},
	}))
```

I tested the version 0.8.0 same error
I also tested using bluetoothctl with -experimental flag

Martim-Sensefinity, 2024-04-17:
[Issue 46](https://github.com/tinygo-org/bluetooth/issues/46) looks similar error

Martim-Sensefinity, 2024-04-17:
I downgraded bluetoothctl to 5.66 and **works**.

The version number is suggested in: [bluez.org/download](https://www.bluez.org/download/)
I followed the following tutorial: [askubuntu.com](https://askubuntu.com/questions/612295/how-to-upgrade-to-the-latest-bluez-version-on-ubuntu-12-04-lts)

Reboot was required.

aykevl, 2024-04-27:
I can confirm this is an issue with BlueZ 5.75 on Fedora 39. It appears to be an upstream bug, so I'll wait until they've investigated the issue.
Thanks @webcaptcha for pointing to the upstream bug report!

aykevl, 2024-05-03:
Looks like the bug has been found and a fix submitted, so hopefully this will be fixed in the next BlueZ version.
(In the meantime I hope we can find a workaround for this bug!)

aykevl, 2024-05-25:
This is fixed in BlueZ 5.75 (tested on Fedora 40). I don't think there's anything to fix here from our side, so closing.

closed 2024-05-25 by aykevl