tinygo-org/bluetooth #272: RSSI naming

elgohr, 2024-05-21:
Is the description of https://github.com/tinygo-org/bluetooth/blob/12b6f0bc25b665a683fd3143cfaef76dadfb619b/gap.go#L109 correctly?
In other contexts RSSI stands for `Received Signal Strength Indicator`, which would be the strength of the signal received.
When I see it correctly this is also what is implemented - isn't it?

aykevl, 2024-05-21:
The comment is perhaps a bit confusing. Every incoming advertisement packet can have a different RSSI value (due to noise, change in distance, people moving around, etc) so this value the RSSI value from the last received packet.
Perhaps a better comment would be:

```go
// The RSSI value of the received advertisement packet.
```

elgohr, 2024-05-21:
I would go for `Signal strength of the  advertisement packet.`.

Otherwise we explain RSSI with RSSI...

closed 2024-05-21 by aykevl