tinygo-org/bluetooth #284: How to select a secondary adapter?

evilsocket, 2024-08-14:
Hi, I've tried to look into the code and the other open issues but I couldn't find a solution to this. Some machines might have multiple bluetooth adapters, however every example only uses bluetooth.DefaultAdapter. 

I could not find a way to select another adapter in the system.

larkwiot, 2024-12-03:
Can you create an Adapter struct with the id field set to the device you want?

evilsocket, 2024-12-03:
Not on macOS https://github.com/tinygo-org/bluetooth/blob/release/adapter_darwin.go#L12

deadprogram, 2025-01-10:
This https://github.com/tinygo-org/bluetooth/pull/303 added ability to choose adapter on Linux.

Still no way to do this on any other platforms.

evilsocket, 2025-02-24:
understood, i guess it makes sense since you can't use external adapters on macOS

deadprogram, 2025-03-05:
https://github.com/tinygo-org/bluetooth/pull/303 allows this for Linux. However macOS is very restrictive.

Seems like the question has been answered. Can this issue be closed now?

ansoni-san, 2025-03-05:
> understood, i guess it makes sense since you can't use external adapters on macOS

You can use external
adapters on macOS (the OS itself will switch which one is the main one). It just requires plugging the adapter in and rebooting or logging in and logging out. 

Although if you wanted to control which adapter is used directly and potentially allow yourself to use a different one than the current OS-selected adapter you would need to interact with the device at the HCI level, which might be outside of the scope of this library (most projects I knew of that did this have since dropped that approach or are unmaintained).