tinygo-org/bluetooth #290: ConnectHandler not handled

gitchander, 2024-08-29:
After using SetConnectHandler events are not handled.

BenjaminMickler, 2024-11-02:
I have the same problem

movsb, 2024-12-01:
dup #256

deadprogram, 2025-01-30:
Now closing as part of release `v0.11.0`. Thank you!

closed 2025-01-30 by deadprogram

movsb, 2025-04-01:
@deadprogram It seemed that for Linux, the device still won't be notified when being connected, only when in disconnected.

https://github.com/tinygo-org/bluetooth/blob/b82048cd9da0fdabb6f2508f461c364184087b3a/gap_linux.go#L394-L406

deadprogram, 2025-04-01:
@movsb I think that on connection this will be called here:
https://github.com/tinygo-org/bluetooth/blob/dev/gap_linux.go#L450

and on disconnection this will be called here:
https://github.com/tinygo-org/bluetooth/blob/dev/gap_linux.go#L460

That is calling the handled on both connect/disconnect, is it not?

movsb, 2025-04-01:
@deadprogram Thanks for the reply. 

I'm connecting a Linux device from a Host computer.
For the second parameter of the ConnectHandler, it is:

* `true` for `Adapter.Connect`, which is called from the Host computer;

  This method itself calls the ConnectHandler just before returning and returns the connected device. It seems not very useful.

* `false` for `device.Disconnect`, which is **also** called from the Host computer.

For the Device itself, there's no way`of knowing when it's being connected or disconnected. Am I right?