tinygo-org/bluetooth #300: bug: HCI-based adaptors broken when acting as peripheral

deadprogram, 2024-10-28:
Since commit 457af7571ab5e7c3a9338cdd901a3b23b3c302d0 is appears that the HCI-based adaptors are broken. They advertise without a problem, however on connecting to them they do not correctly respond as follows:

- cyw43439 is able to respond correctly to requests to describe services/characteristics. However, it cannot send notifications correctly.

- ninafw is not able to respond correctly to requests to describe services/characteristics.

jsoriano, 2024-11-17:
Hey @deadprogram,

I was trying to use this library with an `arduino-nano33` (Arduino Nano 33 IoT), and `adapter.Enable()` already fails. I can confirm the issue starts happening on https://github.com/tinygo-org/bluetooth/commit/457af7571ab5e7c3a9338cdd901a3b23b3c302d0.

This line looks suspicious to me:
https://github.com/tinygo-org/bluetooth/commit/457af7571ab5e7c3a9338cdd901a3b23b3c302d0#diff-1acd23db7ad68fc2a873aff592c71288609ca6c78895f3c3ac3249791a1e2ae8R19
```
c := sz + 4 - (sz % 4)
```
When `sz` is 4, `c` will be 8, is this intended? Could it be reading parts of the next event when `sz` is already a multiple of 4?

I have tried to modify this line so it keeps multiples of 4 as they are, and it seems to work better.
```
c := sz + (4 - (sz % 4)) % 4
```

What is the intention of this line? To get the closest multiple of 4, or the next one?

Would it make sense to revert to the previous code based on `ReadByte()`?

deadprogram, 2024-11-17:
Hi @jsoriano thanks for the correction, I added it to PR #304 just now.

deadprogram, 2025-01-30:
Now closing as part of release `v0.11.0`. Thank you!

closed 2025-01-30 by deadprogram