tinygo-org/bluetooth #316: UUID API is confused

kortschak, 2025-01-01:
The UUID API makes use of (visible) uint32 backing data, this is not currently endian-agnostic and can be interacted with by users directly.

- [ ] bytes representation of the UUID does not roundtrip: https://go.dev/play/p/bccbz_dNHAO: suggest maintaining order.
- [ ] uint32 elements can be interacted with by client code: https://go.dev/play/p/XZ6wRHlti3N: suggest making the type `struct { [4]uint32 }` if the uint32 approach is retained.
- [ ] endianness of data is mixed machine-endian and little-endian: this is only safe on little-endian architecture,

kortschak, 2025-01-14:
Thanks, @aykevl. Yeah, for the last one, I think you are correct. I spent some time yesterday trying to prove my intuitions and could not. I'll strike it.

aykevl, 2025-09-24:
Feel free to make a PR to hide these implementation details :)

kortschak, 2025-09-25:
The use of reversed order bytes looks to be completely entrenched in the API and its use, so I'm not sure what to do about that. As you say, we could unexport the `Bytes` method, but we can't replace it with the expected order, because that would cause additional confusion since the order would reverse. I thought about adding `BytesBigEndian`, but that's hideous.

acouvreur, 2026-10-17:
Fixed by #384.

closed 2026-10-17 by acouvreur