tinygo-org/bluetooth #336: UUID String() method not accessible?

ssokol, 2025-01-20:
Using version 0.10.0, the following line:

```golang
fmt.Printf("Subscribing to notifications on characteristic %s...\n", targetChar.UUID.String())
```

causes the following error:

```
./fvFlightBar.go:76:87: targetChar.UUID.String undefined (type func() bluetooth.UUID has no field or method String)
```

Both the docs and the code seem to think there's a String() method but the compiler begs to differ.

For more context, here's more of the code:

```golang
	// 6. Discover the characteristic within that service
	chars, err := targetService.DiscoverCharacteristics([]bluetooth.UUID{charUUID})
	if err != nil {
		log.Fatalf("Failed to discover characteristics: %v", err)
	}
	if len(chars) == 0 {
		log.Fatalf("Characteristic %s not found", charUUID.String())
	}
	targetChar := chars[0]

	// 7. Subscribe to notifications on the found characteristic
	fmt.Printf("Subscribing to notifications on characteristic %s...\n", targetChar.UUID.String())
	err = targetChar.EnableNotifications(func(value []byte) {
		// Convert received bytes to a hexadecimal string
		hexData := hex.EncodeToString(value)
		fmt.Printf("Notification data (hex): %s\n", hexData)
	})
	if err != nil {
		log.Fatalf("Failed to subscribe to notifications: %v", err)
	}
```

deadprogram, 2025-01-21:
What is the target OS you  are trying to run on?

kortschak, 2025-01-22:
@ssokol You need to call `.UUID` on `DeviceCharacteristic`; it does not have a `UUID` field, it has a method (type *func() bluetooth.UUID* has no field or method String).

deadprogram, 2025-05-04:
Closing since it appears this question was answered. Thanks everyone!

closed 2025-05-04 by deadprogram