tinygo-org/bluetooth #340: [Linux] Scanning in progres gets out of sync if starting to scan while adapter is powered off

Lenart12, 2025-02-19:
As in the title, when trying to start scan but adapter is powered off it will internaly think scan is active but that is not the case.

Minimal reproduction:
```go
//go:build linux
// +build linux

package bluetooth

import (
	"testing"
	"time"

	"github.com/godbus/dbus/v5"
)

func (a *Adapter) SetPowered(powered bool) error {
	return a.adapter.SetProperty("org.bluez.Adapter1.Powered", dbus.MakeVariant(powered))
}

func TestScanWhilePoweredOff(t *testing.T) {
	adapter := DefaultAdapter
	if err := adapter.Enable(); err != nil {
		t.Skipf("skipping test: failed to enable adapter: %v", err)
	}

	if err := adapter.SetPowered(false); err != nil {
		t.Fatalf("failed to power off adapter: %v", err)
	}

	scanFinished := make(chan struct{})
	errCh := make(chan error)
	go func() {
		defer close(scanFinished)
		t.Log("starting scan")
		if err := adapter.Scan(func(_ *Adapter, _ ScanResult) {
			// Do nothing.
		}); err != nil {
			errCh <- err
		}
	}()

	// Wait for the scan to start.
	select {
	case err := <-errCh:
		// expected error
		if err.Error() != "Resource Not Ready" {
			t.Fatalf("unexpected error: %v", err)
		}
	case <-time.After(1 * time.Second):
		t.Fatal("scan did not finish")
	case <-scanFinished:
		t.Fatal("scan finished without error")
	}

	if err := adapter.SetPowered(true); err != nil {
		t.Fatalf("failed to power on adapter: %v", err)
	}

	// Wait for adapter to be ready.
	time.Sleep(1 * time.Second)

	// Start scan, again.
	if err := adapter.Scan(func(_ *Adapter, _ ScanResult) {
		// Do nothing.
	}); err != nil {
		t.Fatalf("failed to restart scan: %v", err)
	}
	t.Log("pass: scan started successfully")
}
```

deadprogram, 2025-02-20:
@Lenart12 what do you think should happen in this case? Some possibilities:

- calling `adaptor.Enable()` on Linux automatically calls something like `SetPowered()`
- calling `adaptor.Enable()` on Linux checks to see if the adaptor is powered, and if not returns an error.
- something else?

deadprogram, 2025-02-20:
OK, I think I misunderstood the original question. My bad.

My current understanding is that if the adaptor is powered down while already scanning, you want to return right away from scan with an error.

If that is the case, then this is probably the place to try to catch that event happening:
https://github.com/tinygo-org/bluetooth/blob/release/gap_linux.go#L232

deadprogram, 2025-02-20:
@Lenart12 see https://github.com/tinygo-org/bluetooth/pull/342

deadprogram, 2025-02-20:
When I live tested on my local machine, and powered off the Bluetooth adaptor while scanning with this branch:

```shell
found device: *****************                                                             
found device: *****************                                                             
found device: *****************                                                             
found device: *****************                                                             
found device: *****************                                                             
found device: *****************                                                             
found device: *****************                                                                                                                     
panic: failed to start scan: bluetooth: adaptor is not powered                     
```

closed 2025-02-25 by Lenart12