tinygo-org/bluetooth #349: Advertisement ManufacturerData not sent on Windows

nvx, 2025-02-24:
Calling `Advertisement.Configure` with `ManufacturerData` set on Windows followed by `Advertisement.Start()` does not seem to affect the manufacturer data sent over the air when looking at the advertisements in the nRF Connect app. I can see the advertisement ServiceUUID fine but no manufacturer specific data appears.

I don't know enough about winrt to be able to diagnose further, but if there's any instructions around how to further debug this I'd be happy to.

nvx, 2025-02-24:
After some more testing it looks like what's happening is there are two separate advertisements with different mac addresses, one with the service uuid and device name, while the other only has the manufacturer specific data. Is there some linkage between the publisher and service that's missing?

deadprogram, 2025-02-24:
Is this the same as https://github.com/tinygo-org/bluetooth/issues/302

nvx, 2025-02-24:
> Is this the same as [#302](https://github.com/tinygo-org/bluetooth/issues/302)

I initially thought the same, but that one is when scanning while this one is about advertising.