tinygo-org/bluetooth #352: v0.11: Windows: Indicate not supported

brentyates, 2025-03-19:
On windows, I'm getting the message "Indicate not supported" when I try to enable notifications. My app works perfect in version 0.10 though, and it works perfectly on Mac.

I believe the issue is found in this PR: https://github.com/tinygo-org/bluetooth/pull/293.

I would be willing to create a PR, but I would like someone to confirm... is the logic backwards on this line: https://github.com/tinygo-org/bluetooth/pull/293/files#diff-e1b3874210bb210694f1b08b334139427db8fe34db0a573d298ca0f701094781R377.

I can do more investigation to determine if my characteristic is actually using notify or indicate in this scenario. Without further investigation though, I think the logic should be checking if the property is non zero instead of zero, and same with the line 2 lines further down, it should again be checking if the GattCharacteristicPropertiesIndicate is non zero and using indicate if it is?

I will follow up with more debugging later this week.

deadprogram, 2025-03-23:
Perhaps @jagobagascon can clarify this?

jagobagascon, 2025-03-23:
Yes, 100% we should be checking for a non-zero value. This is the `if` before that change:

https://github.com/tinygo-org/bluetooth/blob/8503b6ba1240fb47d4a144aa1ea6368f17ee539e/gattc_windows.go#L408-L412

deadprogram, 2025-05-04:
Labeled to close this issue after next release.

PR #359 (2025-05-04): windows: Fix Notify/Indicate characteristics detection
As per https://github.com/tinygo-org/bluetooth/issues/352#issuecomment-2746085930, fixes #352.
closed 2025-08-02 by deadprogram