tinygo-org/bluetooth #354: Problem with SetConnectedHandler() on Raspberry Pi OS Peripheral Device

kevin-activ, 2025-04-10:
Same issue as https://github.com/tinygo-org/bluetooth/issues/145
That was closed but it's still an issue. 

Unfortunately this is still not working with v0.11.0 on Raspberry Pi Linux

I'm running this lib version:
` Dep: &{Path:tinygo.org/x/bluetooth Version:v0.11.0 Sum:h1:32ludjNnqz6RyVRpmw2qgod7NvDePbBTWXkJm6jj4cg= Replace:<nil>}`

and the latest version of Pi OS
`Linux pi4work 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux`

I have my program running as a BLE peripheral and the connections work just fine and I can send data, but the connect handler is never called. No matter where I call SetConnectHandler() from. I've tried before and after enabling the adapter, before and after starting the advertising, etc 
```
adapter.SetConnectHandler(func(d bluetooth.Device, c bool) {
		println("handler")
		connected = c
	})
```

feliks912, 2025-04-16:
connectHander, as per code in this repo as of 0.11.0, gets properly handled only on cyw43439

https://github.com/tinygo-org/bluetooth/blob/b82048cd9da0fdabb6f2508f461c364184087b3a/gap_hci.go#L497

and the gap_linux.go source file, one which Linux uses, handles connectHandler only when connection or disconnection is called from the Go program itself. It doesn't 'catch' the bluetooth connection made from elsewhere (feature request...)

Instead, the following is a Google Gemini 2.5 Pro's dbus handler which will show the changes in connected devices:

`package main

import (
	"fmt"
	"log"
	"os"
	"os/signal"
	"strings" // Import strings package
	"syscall"

	"github.com/godbus/dbus/v5"
)

const (
	bluezServiceName        = "org.bluez"
	objectManagerInterface  = "org.freedesktop.DBus.ObjectManager" // Correct interface for GetManagedObjects
	dbusPropertiesInterface = "org.freedesktop.DBus.Properties"    // Standard properties interface
	adapterInterface        = "org.bluez.Adapter1"
	deviceInterface         = "org.bluez.Device1"
)

type Address struct {
	MAC string
}

type Device struct {
	Address Address
	Path    dbus.ObjectPath // Store the path for easier lookup
	device  dbus.BusObject
}

func main() {
	conn, err := dbus.SystemBus()
	if err != nil {
		log.Fatalf("Failed to connect to system bus: %v", err)
	}
	defer conn.Close()

	// Get the ObjectManager interface on the root BlueZ object
	objManager := conn.Object(bluezServiceName, "/")

	// --- Corrected GetManagedObjects Call ---
	var managedObjects map[dbus.ObjectPath]map[string]map[string]dbus.Variant
	err = objManager.Call(objectManagerInterface+".GetManagedObjects", 0).Store(&managedObjects)
	if err != nil {
		// Add more context to the error log
		log.Fatalf("Failed to get managed objects from %s: %v", objectManagerInterface, err)
	}
	// --- End Correction ---

	// Find the first Bluetooth adapter
	var adapterPath dbus.ObjectPath
	for path, interfaces := range managedObjects {
		if _, ok := interfaces[adapterInterface]; ok {
			// Ensure it's a valid adapter path (e.g., /org/bluez/hci0)
			if strings.HasPrefix(string(path), "/org/bluez/hci") {
				adapterPath = path
				log.Printf("Found Bluetooth adapter: %s", adapterPath)
				break // Use the first adapter found
			}
		}
	}
[... 101 lines cut ...]
								connectedDevices[address] = device
								fmt.Printf("Device connected:   MAC=%s, Path=%s\n", address, path)
							} else {
								// Already known as connected, maybe properties updated? Log if needed.
								// log.Printf("Device %s (%s) already connected, properties updated.", address, path)
							}
						} else {
							// Device disconnected
							if _, exists := connectedDevices[address]; exists {
								delete(connectedDevices, address)
								fmt.Printf("Device disconnected: MAC=%s, Path=%s\n", address, path)
							} else {
								// We didn't know it was connected. Might happen on startup or if missed signal.
								// log.Printf("Device %s (%s) disconnected, but wasn't tracked as connected.", address, path)
							}
						}
					}
				}
			}
		}
	}
}
`

Since it works with dbus it should work with raspberry pi. Tested on Linux Mint. I will edit this when I make it work as a simple function, equal to the connect / disconnect handler in this Go package.

deadprogram, 2025-06-13:
Labeling to close on next release, since this has been addressed by #369